bitcraze/crazyflie-lib-python #408: Streaming Ranging Distance Info from the Localization Class Using the Python Library

brianwu568, 2023-07-21:
Hello Developer Team,

I am currently using the python library to stream data from and control a drone, and one of the objectives that I am interested in involves the Localization class, defined in cflib/crazyflie/localization.py. I currently have some loco positioning nodes, and upon inspection of the class it appears that the function _incoming() will be most useful. I believe that the type of packet I need to stream distance info from the loco nodes is "RANGE_STREAM_REPORT", and lines 102-102 should give the required anchor and distance info.

`               anchor_id, distance = struct.unpack('<Bf', raw_data[:5])
                decoded_data[anchor_id] = distance
`

Now, in the main SyncCrazyFlie context manager, where I have defined a CrazyFlie object cf, I added the following to configure the packet sender and receiver:

` # Configure Packet Sender and Receiver
    localization = controller.cf.loc
    localization.receivedLocationPacket.add_callback(localization_callback) # Register the localization_callback as a listener to the receivedLocationPacket event of the Localization class
    localization._cf.add_port_callback(CRTPPort.LOCALIZATION, localization._incoming) # Start the data communication with the Crazyflie by adding the Localization port callback
    `

The function localization_callback is custom defined, and takes the following form:

`def localization_callback(packet):
    # Access and process the received localization data
    if packet.type == Localization.RANGE_STREAM_REPORT:
        decoded_data = packet.data
        anchor_id, distance = struct.unpack('<Bf', decoded_data[:5])
        # Process the anchor_id and distance data here
        print(f"Received range data from anchor {anchor_id}: {distance} meters")`

However, with this implementation I am never getting a single print statement in the output, which suggests to me that the callback function is never being called. What would be really useful to me is within the SyncCrazyFlie context manager, I could call on the callback function at a specific point, and print out the distance data from the packets at that point in time. Any insights and help on this issue would be greatly appreciated.

ataffanel, 2023-07-21:
It is not enough to add a callback, you also need to enable the range streaming in the Crazyflie. This is done by setting the parameter "locSrv.enRangeStreamFP32" to 1: https://github.com/bitcraze/crazyflie-firmware/blob/e56b7fddf6e6ac9e10d9bacfffb4c4bfcd629219/src/modules/src/crtp_localization_service.c#L463-L465

If you do that after setting up the callback, you should start receiving measurements from the Crazyflie.

brianwu568, 2023-07-21:
Thanks for the comment, your fix resolved the problem and I am now receiving distance packets from the LPS. I will close this issue now.

closed 2023-07-21 by brianwu568