bitcraze/crazyflie-lib-python #412: Why it reminded me "Could not add Measurement log config, bad configuration"，when I added a new log variable ?

czyv587, 2023-08-16:
I added the 3d tof driver in the crazyflie-firmware, I can see the log option on the crazy client, and I can also draw the curve, but adding the 3d tof log variable in crazyflie-lib-python\examples\multiranger\multiranger_pointcloud.py reminds me "Could not add Measurement log config, bad configuration". What is the reason? How should I modify it?Here is my code：
`lmeas = LogConfig(name='Meas', period_in_ms=100)
        # lmeas.add_variable('range.front')
        # lmeas.add_variable('range.back')
        # lmeas.add_variable('range.up')
        # lmeas.add_variable('range.left')
        # lmeas.add_variable('range.right')
        # lmeas.add_variable('range.zrange')
        lmeas.add_variable('stabilizer.roll')
        lmeas.add_variable('stabilizer.pitch')
        lmeas.add_variable('stabilizer.yaw')
        lmeas.add_variable('tof.distance0', 'float')
        lmeas.add_variable('tof.distance2', 'float')
        lmeas.add_variable('tof.distance5', 'float')
        lmeas.add_variable('tof.distance6', 'float')
        lmeas.add_variable('tof.distance16', 'float')
        lmeas.add_variable('tof.distance32', 'float')
        lmeas.add_variable('tof.distance48', 'float')
        lmeas.add_variable('tof.distance57', 'float')
        lmeas.add_variable('tof.distance59', 'float')
        lmeas.add_variable('tof.distance61', 'float')
        lmeas.add_variable('tof.distance63', 'float')
        lmeas.add_variable('tof.distance47', 'float')
        lmeas.add_variable('tof.distance31', 'float')
        lmeas.add_variable('tof.distance15', 'float')

        try:
            self.cf.log.add_config(lmeas)
            lmeas.data_received_cb.add_callback(self.meas_data)
            lmeas.start()
        except KeyError as e:
            print('Could not start log configuration,'
                  '{} not found in TOC'.format(str(e)))
        except AttributeError:
            print('Could not add Measurement log config, bad configuration.')`
[image]
[image]
[image]

knmcguire, 2023-08-16:
Hi! You are probably adding too many floats to the log config. As you already noticed in the cfclient, there is a limit on how many you can add, which are 26 bytes: https://www.bitcraze.io/documentation/repository/crazyflie-lib-python/master/user-guides/python_api/#variables-and-logging

czyv587, 2023-08-16:
> Hi! You are probably adding too many floats to the log config. As you already noticed in the cfclient, there is a limit on how many you can add, which are 26 bytes: https://www.bitcraze.io/documentation/repository/crazyflie-lib-python/master/user-guides/python_api/#variables-and-logging
Thank you very much！！ your answer helped me to solve the problem！！

closed 2023-08-24 by krichardsson