bitcraze/crazyflie-lib-python #421: QT6 needs extra dependencies

knmcguire, 2023-10-17:
Currently we have in our documentation that people have to install an xcb library among other things:

`sudo apt install git python3-pip libxcb-xinerama0```

Now that we have upgraded the cfclient to QT6.5, I got an error on my Ubuntu 22.04 (python 3.10) that it still needed an additional library.

```
INFO:__main__:Disabling STL printouts
INFO:cfclient.utils.input.inputreaders:Input readers: ['linuxjsdev', 'pysdl2']
INFO:cfclient.utils.input.inputreaders:Successfully initialized [linuxjsdev]
INFO:cfclient.utils.input.inputreaders:Could not initialize [pysdl2]: No SDL2 support on Linux
INFO:cfclient.utils.input.inputinterfaces:Found interfaces: ['leapmotion', 'wiimote', 'zmqpull']
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [leapmotion]: Leap Motion library probably not installed (No module named 'leapsdk')
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [wiimote]: Missing cwiid (wiimote) driver No module named 'cwiid'
INFO:cfclient.utils.config:Dist config read from /home/kimberly/Development/bitcraze/python/crazyflie-clients-python/src/cfclient/configs/config.json
INFO:cfclient.utils.config:Config file read from [/home/kimberly/.config/cfclient/config.json]
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [zmqpull]: ZMQ input disabled in config file
INFO:cfclient.utils.zmq_led_driver:ZMQ led disabled in config file
INFO:cfclient.utils.zmq_param:ZMQ param disabled in config file
INFO:__main__:If startup fails because of "xcb", install dependency with `sudo apt install libxcb-xinerama0`.
qt.qpa.plugin: From 6.5.0, xcb-cursor0 or libxcb-cursor0 is needed to load the Qt xcb platform plugin.
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: minimalegl, eglfs, vkkhrdisplay, vnc, linuxfb, wayland, xcb, minimal, wayland-egl, offscreen.

```

This was solved by installing 
```
sudo apt-get install libxcb-cursor0
```

The question is, is this necessary for every machine or just mine? Not everyone has this problem but if it is universal than we need to add this to documentation and the error message.

tobbeanton, 2023-10-17:
I also had to do this but I think @krichardsson did not have to and I think the difference is a newer Ubuntu version.

knmcguire, 2023-10-18:
This was a good recourse from kristoffer as well: https://doc.qt.io/qt-6/linux-requirements.html

knmcguire, 2023-10-24:
wait... let me transfer this to the cfclient as this is not a cflib problem...