bitcraze/crazyflie-lib-python #446: Not able to find Crazyradio Dongle on MacOS with Python 3.12 - Exception: Cannot find a Crazyradio Dongle

skovsen, 2024-04-02:
Environment:
- MacOS 14.3.1
- Python 3.12.0
- cflib 0.1.25.1

Using Python 3.12, I keep getting the error: Exception: Cannot find a Crazyradio Dongle
when trying to connect to a Crazyflie.
If I downgrade Python til 3.11.4 it works fine.

I have seen this, both when using CFClient as well as when developing my own code

knmcguire, 2024-04-03:
Thanks for reporting this! this is an weird error so we will try to investigate it soon. 3.12 should still be supported for mac

skovsen, 2024-04-03:
Hi.
I agree.
To clarify, I'm using a Crazyradio 2.0. I forgot to add that into the original post.

But, just to ensure myself, I tried once more, with the following steps:
Not working with python 3.12.0
`python -m venv env && source env/bin/activate && pip install --upgrade pip && pip install cfclient`
`cfclient`

Which results in the following output in my terminal:

`WARNING:cfclient.ui.main:Cannot find a Crazyradio Dongle`
`Cannot find a Crazyradio Dongle`
`INFO:cflib.drivers.cfusb:Looking for devices....`
`WARNING:cflib.crtp.usbdriver:Exception while scanning for Crazyflie USB: No backend available`

And I'm not able to connect to a CF.

If I change my python over to 3.11.4 and try again, like this:
` pyenv local 3.11.4`
`python -m venv env && source env/bin/activate && pip install --upgrade pip && pip install cfclient`
`cfclient`

I works fine.

ataffanel, 2024-04-03:
To get some more info on how to reproduce the problem: How do you install Python? Using the official installer or a package manager like brew/port?

skovsen, 2024-04-03:
I'm using pyenv

ToveRumar, 2024-04-04:
Hi!
I have been able to reproduce  it and it  looks like there might be some issue connecting the backend usb lib to pyUSB. However, python 3.12 is not officially supported (see https://github.com/bitcraze/crazyflie-lib-python/issues/425). 

We hope to fix this soon.

skovsen, 2024-04-09:
Hi.
I think it is not completely the same, as the issue referenced in #425 is a build time issue, while here it does build the system. This issue looks to stem from pyUSB in 3.12 does not support the Crazyradio. I'm not an expert when we come this close to the hardware, but it looks a bit of a different issue.
Hopefully, it will be solved at the same time, though :)

ToveRumar, 2024-04-10:
Hi!

I agree, it looks a bit different. But we hope to solve it all in one go. 
Thanks for reporting!

ToveRumar, 2024-09-05:
@skovsen I just had some time to look at this issue and I solved it on an M1 Mac by installing libusb using homebrew:
`brew install libusb` 

I have also seen some people needing to link in the lib path like this:
`export DYLD_LIBRARY_PATH="YOUR_HOMEBREW_PATH/lib:$DYLD_LIBRARY_PATH"`
If this step is needed or not  seems to depend on where your homebrew installation is located. 

Could you try this out and see if it solves your issue as well?

skovsen, 2024-09-20:
> @skovsen I just had some time to look at this issue and I solved it on an M1 Mac by installing libusb using homebrew: `brew install libusb`
>
> I have also seen some people needing to link in the lib [...]

Hi @ToveRumar 

I have tested this, and I can confirm that it works.
I had to add the DYLD_LIBRARY_PATH env step as well.
I have tested this with libusb version: 1.0.27

ToveRumar, 2024-09-23:
Hi @skovsen!

Great, thank you for testing this!
I'm closing this issue now then, since its solved. 

BR
Tove

closed 2024-09-23 by ToveRumar