bitcraze/crazyflie-lib-python #467: Investigate communication speed

knmcguire, 2024-09-03:
We have been saying that the communication speed (packets per second) should be about 400ish packets per second. The maximum size is 26 bytes (without the header). The minimum period per packet is 10ms

Let's check this with the current situation! There might be some bottlenecks here and there (usb to crazyflie), but it would be nice to have some ice cold hard facts in numbers to see what the current situation is.

whoenig, 2024-09-04:
FYI, in Crazyswarm2 in the cpp backend we continuously monitor these sort of connection statistics and also allow for user-provided warning threshold.

knmcguire, 2024-09-04:
Thanks! Yes I was aware that you have implemented something like that. Eventually we want to add more quality checks to the cflib as well, but for now this is just investigating without adding a new functionality, so a separate issue will be made for the quality stats (for better user understanding of what is happening and for CI)

This was the PR where you implemented the latests on these quality stats right, in cpp link? https://github.com/bitcraze/crazyflie-link-cpp/pull/28

whoenig, 2024-09-04:
The statistics in the link had been around for a long time (other backends might have it too?). It was hooked up to the warning system here: https://github.com/IMRCLab/crazyswarm2/commit/d71ee98ad6306dd6a9bd8da4c2aad38235618354.

knmcguire, 2024-09-12:
I'll close this as we have determined that there is not really that much wrong with the overal communication packet sending but many other things that we simply can't monitor. After we have implemented #469 we can make any issues that we see arising of our tests to specific issues here, but at least the investigation has lead us to the path of better comm stats improvements 😄

closed 2024-09-12 by knmcguire