bitcraze/crazyflie-lib-python #472: Add broadcasting

knmcguire, 2024-09-11:
[Crazyflie-link-cpp](https://github.com/bitcraze/crazyflie-link-cpp/tree/master) can do swarm broadcasting, so why not the link on the cflib! 

This is a missing functionality in Crazyswarm2 that blocks the cflib from being a drop in replacement for the cpp backend: https://github.com/IMRCLab/crazyswarm2/issues/423

ArisMorgens, 2025-03-10:
Hello Nagesh!

It is so nice to hear that you are interested in contributing!
Swarm broadcasting is indeed not supported by the cflib and is only implemented in the firmware.

Our most recent use of broadcasting was for the [decentralized brushless swarm demo](https://www.bitcraze.io/2025/03/decentralized-brushless-swarm-demo/). For that, we used a "hacky" solution where a non-flying Crazyflie is responsible to broadcast to all others. 
You can find the [source code](https://github.com/bitcraze/crazyflie-firmware-experimental/tree/arena-demo/examples/demos/decentralized_swarm) for the demo in our [crazyflie-firmware-experimental](https://github.com/bitcraze/crazyflie-firmware-experimental) repository.
Let us know if you need any further help.

NageshEranki, 2025-03-27:
Hi Aris,

Sorry for the delay in getting back to you. Thanks for the solution! There are no such things as 'hacky' solutions :) While it's a great starting point, I'm more interested in extending `crazyflie-lib-python` to do what `crazyflie-link-cpp` does.

Between my message on this thread and now, I used crazyflie-link-cpp directly to broadcast position feedback from mocap and position setpoints to  5 Crazyflies. Attached is an example script for the same. 

I flash an app on the drone; I attached the C file for that too.

Do you think it is possible to directly program in python what `crazyflie-link-cpp` does?

[app_broadcast_fixed_point.c.txt](https://github.com/user-attachments/files/19494694/app_broadcast_fixed_point.c.txt)

[broadcast_fixed_point_poses.py.txt](https://github.com/user-attachments/files/19494688/broadcast_fixed_point_poses.py.txt)

NageshEranki, 2025-04-03:
Hi Tove!

Thanks a lot for the pointers! Let me see what I can do