bitcraze/crazyflie-lib-python #478: NixOS

shot-codes, 2024-09-26:
I'm leaving this here for anyone who is attempting to run `cfclient` on NixOS. The following `shell.nix` allows for successful launch of the application. I haven't done extensive testing but will try to come back here with any relevant updates if I come across something.

> The shell assumes nix-ld is installed (` programs.nix-ld.enable = true;`)

```nix
with import <nixpkgs> {};
mkShell {
  NIX_LD_LIBRARY_PATH = lib.makeLibraryPath [
    stdenv.cc.cc
    zlib
    libGL
    libxkbcommon
    kdePackages.full
    kdePackages.wayland
    fontconfig
    libusb1
  ];
  NIX_LD = lib.fileContents "${stdenv.cc}/nix-support/dynamic-linker";
  buildInputs = [ python312 ];
  shellHook = ''
    export LD_LIBRARY_PATH=$NIX_LD_LIBRARY_PATH
    source .venv/bin/activate
  '';
}
```

gemenerik, 2024-09-27:
Thanks for the contribution!

mettz, 2025-06-15:
@shot-codes @gemenerik after the approval of https://github.com/NixOS/nixpkgs/pull/410846 this should become as easy as this:
```
with import <nixpkgs> {};
mkShell {
  buildInputs = [ cfclient ];
}
```

ataffanel, 2025-06-18:
This is a great news! Thanks @brianmcgillion for adding it to Nix and thanks @mettz for the update here!

I think we should add to the documentation that this is a possibility (an maybe adding a .nix file in there as well?)

brianmcgillion, 2025-06-18:
There are a few ways to support this with nix, I will list a few.

1. with a `flake.nix`. but could be a bit of work to maintain.
2. with a `shell.nix` as listed above. the only issue that I can see is if you add a new dependency to the pyproject.toml you might have to update the nixpkgs for that to reflect in the build inputs to the shell.
3. `nix develop nixpkgs#cfclient` - similar to the `shell.nix ` it will create a dev environment in the current directory with all the development requirements in place, no need to add anything to the repo. Same caveat as `shell.nix` it takes the dependencies from the current released version in nixpkgs.
4. with `devenv`

I might propose the 4th option as the least maintenance approach, while keeping in sync with development. The project is still managed fully through the pyproject with a small wrapper that will handle the virtual environment and nix integration and a good middle ground if you plan to do active development through nix/nixos.

I created a small template repo for another use case recently, which can act as a reference to explain the use. That you can see here, https://github.com/brianmcgillion/python-kickstart.

You would need the 3 files, `devenv.nix` (the entry point, and really only lines 17-34, rest is just some template ...), `devenv.yaml` (where you will get the nix sources for pulling in the package dependencies) and `.envrc` (just to drop you into the devshell when you enter the dir - not essential but nice to have). once in place you will only have to run `devenv update` occasionally to refresh the nixpkgs sources (creates devenv.lock) so should be almost no maintenance cost.

The repo read me has most of the details, i.e. the few other configs that are needed in the dev machine config to enable it. If interested, I could try and find some time to assist with this or help test it.

closed 2026-09-01 by ataffanel