bitcraze/crazyflie-lib-python #508: bluetooth api upgrade

Ethicallearner77, 2025-01-25:
hey I want help I am running cfclient through esp idf and I am facing the error which is this 
This Windows version (6.3.9600) does not support the required Bluetooth API. Consider updating to a more recent Windows (10.0.10586 or above).
and i can't upgrade my window right now and I don't want to use bluetooth api in my project so how can I disabled this thing if anyone have the solution pls let me now I will be very thankful to you if anyone helps me 

[image]

gemenerik, 2025-01-30:
Hi. Can you copy the content of your console into this thread. That may help us pinpoint what is 

Note that your version of Windows is very outdated and is no longer receiving any updates so you may want to upgrade anyhow. But let's try to figure out where the error is coming from.

ToveRumar, 2025-06-26:
Im closing this issue since this seems to me more of a discussions topic. Feel free to open a discussion on this if needed.