bitcraze/crazyflie-lib-python #511: Adaption to latency detection

SimonZhan-code, 2025-02-03:
Hi,

We are robotics researchers working on control applications with delay, using Crazyflies. I came across [this link](https://www.bitcraze.io/tag/rssi/), which directed me to information about latency metrics. However, it seems that the current support for latency measurement is incomplete.

Is there any partial support available that I could use to obtain link quality or latency measurements?

Thank you in advance!

yyf20001230, 2025-02-03:
I'm bumping this thread. As another researcher working with Crazyflies, I'm also very interested in robust latency and link quality measurement capabilities. This is a critical feature for precise control applications, especially in time-sensitive robotic systems.

gemenerik, 2025-02-05:
There's a draft pull request for visualization in the client, which you can check out [here](https://github.com/bitcraze/crazyflie-clients-python/pull/732). Since it's still in the draft stage, the UI hasn’t been given much thought yet, and it might not be particularly useful for research purposes. Instead, I’d recommend focusing on using callbacks. You can find an example of how to add a callback for latency [here](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/link_quality/latency.py). While there aren’t any examples for other statistics at the moment, you can implement callbacks for them in the same way as shown in the latency example.

We’d really appreciate any feedback you might have, especially regarding which statistics you find useful or if there are any that you feel are missing. If you have suggestions on how things could be improved or changed, we’d love to hear your thoughts!

Hope this helps!

closed 2025-02-05 by gemenerik