bitcraze/crazyflie-lib-python #523: Packet Loss and DeprecationWarning

snydergi, 2025-02-25:
Hi all,

I am working toward trying to do some simply autonomous flights with my Crazyflie 2.1+.

I was trying to follow the tutorial here: https://www.bitcraze.io/documentation/tutorials/getting-started-with-stem-drone-bundle/ since I have the Flow Deck V2 installed and want to try to stick with relative positioning. When I follow the steps in the guide, however and run the code, here is my output:

`(cfAutoEnv) gis@gis76:~/Documents/WinterProject$ autoTestFolder/cfAutoEnv/bin/python autoTestFolder/testRun.py`
`/home/gis/Documents/WinterProject/autoTestFolder/cfAutoEnv/lib/python3.12/site-packages/libusb_package/__init__.py:51: DeprecationWarning: is_resource is deprecated. Use files() instead. Refer to https://importlib-resources.readthedocs.io/en/latest/using.html#migrating-from-legacy for migration advice.
  if resources.is_resource(__name__, _LIBRARY_NAME):
/home/gis/Documents/WinterProject/autoTestFolder/cfAutoEnv/lib/python3.12/site-packages/libusb_package/__init__.py:52: DeprecationWarning: path is deprecated. Use files() instead. Refer to https://importlib-resources.readthedocs.io/en/latest/using.html#migrating-from-legacy for migration advice.
  path_resource = resources.path(__name__, _LIBRARY_NAME)
Traceback (most recent call last):
  File "/home/gis/Documents/WinterProject/autoTestFolder/testRun.py", line 24, in <module>
    with SyncCrazyflie(URI) as scf:
  File "/home/gis/Documents/WinterProject/autoTestFolder/cfAutoEnv/lib/python3.12/site-packages/cflib/crazyflie/syncCrazyflie.py", line 119, in __enter__
    self.open_link()
  File "/home/gis/Documents/WinterProject/autoTestFolder/cfAutoEnv/lib/python3.12/site-packages/cflib/crazyflie/syncCrazyflie.py", line 95, in open_link
    raise Exception(self._error_message)
Exception: Too many packets lost`

Test run contains this code, pulled from the tutorial:
```
`"""
This script shows a simple scripted flight path using the MotionCommander class.

Simple example that connects to the crazyflie at `URI` and runs a
sequence. Change the URI variable to your Crazyflie configuration.
"""
import logging
import time

import cflib.crtp
from cflib.crazyflie.syncCrazyflie import SyncCrazyflie
from cflib.positioning.motion_commander import MotionCommander

URI = 'radio://0/80/250K'

# Only output errors from the logging framework
logging.basicConfig(level=logging.ERROR)

if __name__ == '__main__':
    # Initialize the low-level drivers (don't list the debug drivers)
    cflib.crtp.init_drivers(enable_debug_driver=False)

    with SyncCrazyflie(URI) as scf:
        # Arm the Crazyflie
        scf.cf.platform.send_arming_request(True)
        time.sleep(1.0)

        # We take off when the commander is created
        with MotionCommander(scf) as mc:
            print('Taking off!')
            time.sleep(1)

            # There is a set of functions that move a specific distance
            # We can move in all directions
            print('Moving forward 0.5m')
            mc.forward(0.5)
            # Wait a bit
            time.sleep(1)

            print('Moving up 0.2m')
            mc.up(0.2)
            # Wait a bit
            time.sleep(1)

            print('Doing a 270deg circle')
            mc.circle_right(0.5, velocity=0.5, angle_degrees=270)

            print('Moving down 0.2m')
            mc.down(0.2)
            # Wait a bit
            time.sleep(1)

            print('Rolling left 0.2m at 0.6m/s')
            mc.left(0.2, velocity=0.6)
            # Wait a bit
            time.sleep(1)

            print('Moving forward 0.5m')
            mc.forward(0.5)

            # We land when the MotionCommander goes out of scope
            print('Landing!')
`
```

I am running the code in a python venv so that it can detect cflib.

Any help would be greatly greatly appreciated!

ArisMorgens, 2025-02-26:
Hello snydergi!

The Deprecation Warnings you get are due to your virtual environment and not cflib related. They can be easily solved by replacing the deprecated functions (`is_resource` and `path`) with `files()` in the mentioned path. They can also be ignored.

Regarding the Packet Loss error, it seems that your Crazyflie is not connected. Perhaps `URI = 'radio://0/80/250K'` is not the correct URI for your Crazyflie, so try to replace it. You can get the correct address by connecting to the cfclient through USB.

Let us know if this worked or if you need further assistance.

snydergi, 2025-02-26:
Hi thank you for responding! I am still having issues. The deprecation was fixed very easily, thank you. I should have read the error message closer instead of fearing it.

The packet loss, however is not solved. I previously used the cfclient to do remote control flying and the uri was radio://0/20/2M/E7E7E7E7E7, however now when I am in cfclient and I scan, nothing appears in the 'select an interface' like previously. Did these commands do something to make it so I can no longer remote control either? 
```
sudo groupadd plugdev
sudo usermod -a -G plugdev $USER
echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="1915", ATTRS{idProduct}=="7777", MODE="0664", GROUP="plugdev"' | sudo tee /etc/udev/rules.d/99-crazyradio.rules
```

I also plugged the Crazyflie itself into the laptop and nothing appeared in the cfclient. Also in the cfclient, I see this message when I have the radio, my controller, and the crazyflie all plugged in and hit 'Scan':
```
INFO:cflib.crtp.radiodriver:v3.2 dongle with serial 234057074B3D4AC2 found
INFO:cflib.drivers.cfusb:Looking for devices....
INFO:cflib.crtp.prrtdriver:Initialized PRRT driver.
INFO:cfclient.utils.input:Selected MUX: Normal
INFO:cfclient.ui.main:Role of Generic X-Box pad is Device
INFO:cfclient.utils.input.mux:Adding device Generic X-Box pad to MUX Normal
INFO:cflib.crtp.radiodriver:v3.2 dongle with serial 234057074B3D4AC2 found
INFO:cflib.drivers.cfusb:Looking for devices....
INFO:cflib.crtp.prrtdriver:Initialized PRRT driver.
```

I am using a machine running Ubuntu 24.04 if that is relevant.

Thank you again for the help!!!

snydergi, 2025-02-26:
Update!

I tried a second Crazyflie that I had available in the lab (which was actually the one I was using prior to having to switch to a different board) and it connected and ran the code fine with these modifications. 

So, now my issue is that one of my Crazyflies does not connect to the computer in any way that I can tell. Should I open a new discussion for this? Or what additional info can I provide or ideas can I attempt to try to fix the other one?

Thanks!

ArisMorgens, 2025-02-27:
Hello again!

So nice to hear that you managed to run the script with one of your Crazyflies. Yes, I think it is appropriate to open a new discussion about the new issue you are facing and hopefully we will solve it.
I am closing this issue as it is solved.

closed 2025-02-27 by ArisMorgens