bitcraze/crazyflie-lib-python #559: Motion Capture Library Import Error

AfrifaEben7, 2025-08-27:
I would like to use the motion capture example with OptiTrack. 

`> python mocap_hl_commander.py`
`
Traceback (most recent call last):
  File "/Users/eben/Code/crazyflie-lib-python/examples/mocap/mocap_hl_commander.py", line 34, in <module>
    import motioncapture
ModuleNotFoundError: No module named 'motioncapture'
`

I thought maybe I did not install the library well. I reinstalled, but the problem persists. 

Crazyflie 2.0
Running on MacOS Sequoia

AfrifaEben7, 2025-08-28:
I realised the python version I was using was not the version the motioncapture was compiled for. I was using 13.12 and the motioncapture(1.0a4) was compiled for 3.13.

enyanil, 2025-08-29:
Good find! Let us know if you have any other issues.